aff3ct/aff3ct #87: LDPC decoder generates all 0s.

SubtleMuffin, 2020-10-22:
Hi -- I am trying to use a LDPC encoder and decoder chain as a standalone binary, and I encountered the all 0 outputs of LDPC decoder. As mentioned in #53 , although I have fixed the BPSK mapping, it still does not work. Here is my code listing -- 

```c++
// #include "codec.hpp"
#include "aff3ct.hpp"
#include "math.h"

std::string LDPC_H_MATRIX_NAME("../lib/conf/dec/LDPC/MACKAY_504_1008.alist");

auto get_h_qc(std::string file_name) {
  std::ifstream h_file;
  h_file.open(file_name);
  auto h_mat = aff3ct::tools::QC::read(h_file);
  h_file.close();
  return h_mat;
}

auto get_h_alist(std::string file_name) {
  std::ifstream h_file (file_name);
  auto h_mat = aff3ct::tools::AList::read(h_file);
  h_file.close();
  return h_mat;
}

auto ldpc_encode(const int k, const int n, const std::vector<int> &info_bits) {
  // get h matrix
  auto h_mat = get_h_alist(LDPC_H_MATRIX_NAME);
  auto ldpc_encoder = aff3ct::module::Encoder_LDPC_from_IRA<int>(k, n, h_mat, 1);

  // encode
  std::vector<int> encoded_bits(n);
  ldpc_encoder.encode(info_bits, encoded_bits);
  return encoded_bits;
}

auto ldpc_encode_multiple(const int k, const int n,
                          const std::vector<std::vector<int>> &info_bits) {
  // generate h matrix
  auto h_mat = get_h_alist(LDPC_H_MATRIX_NAME);
  auto ldpc_encoder = aff3ct::module::Encoder_LDPC_from_IRA<int>(k, n, h_mat, 1);

  // encode
  std::vector<std::vector<int>> encoded_bits;
  for (auto frame : info_bits) {
    std::vector<int> encoded(n);
    ldpc_encoder.encode(frame, encoded);
    encoded_bits.push_back(encoded);
  }
  return encoded_bits;
}

auto ldpc_decode(const int k, const int n, const std::vector<float> &received) {
  // generate h matrix
  auto h_mat = get_h_alist(LDPC_H_MATRIX_NAME);
  std::vector<unsigned> read_info_bits_pos(k, 0);
  auto ldpc_decoder = aff3ct::module::Decoder_LDPC_BP_flooding_SPA<>(
      k, n, 30, h_mat, read_info_bits_pos);

  // decode
  std::vector<int> decoded_bits(k);
  ldpc_decoder.decode_siho(received, decoded_bits);

  return decoded_bits;
}

auto ldpc_decode_multiple(const int k, const int n,
                          const std::vector<std::vector<float>> &received) {
  // generate h matrix
  auto h_mat = get_h_alist(LDPC_H_MATRIX_NAME);
  std::vector<unsigned> read_info_bits_pos(k, 0);
  auto ldpc_decoder = aff3ct::module::Decoder_LDPC_bit_flipping_OMWBF<>(
      k, n, 30, h_mat, read_info_bits_pos);

  // decode
  std::vector<std::vector<int>> decoded_bits;
  for (auto frame : received) {
    std::vector<int> decoded(k);
    ldpc_decoder.decode_siho(frame, decoded);
    decoded_bits.push_back(decoded);
  }
  return decoded_bits;
}

template <typename T> auto print_vector(std::vector<T> &vec) {
  std::for_each(vec.begin(), vec.end(),
                [&](const auto i) { std::cout << i << ' '; });
  std::cout << std::endl;
}

int main() {
  int k = 504;
  int n = 1008;
  float snr_max = 20;

  // generate information bits
  auto source =
      std::unique_ptr<aff3ct::module::Source_random<>>(new aff3ct::module::Source_random<>(k));
  std::vector<int> info_bits(k);
  source->generate(info_bits);

  // ldpc encode
  auto encoded_bits = ldpc_encode(k, n, info_bits);

  // generate gaussian noise
  const float mean = 0.0;
  const float stddev = std::sqrt(std::pow(10, -snr_max / 10.0));
  std::normal_distribution<float> dist(mean, stddev);
  std::default_random_engine generator;

  // awgn channel
  std::vector<float> received;
  std::transform(
      encoded_bits.begin(), encoded_bits.end(), std::back_inserter(received),
      [&](const int i) -> float { return -i * 2 + 1 + dist(generator); });

  // ldpc decode
  auto decoded_bits = ldpc_decode(k, n, received);

  // print vectors
  std::cout << "--- received symbols  ---" << std::endl;
  print_vector(received);
  std::cout << "--- info bits         ---" << std::endl;
  print_vector(info_bits);
  std::cout << "--- decoded bits      ---" << std::endl;
  print_vector(decoded_bits);

  // check for errors
  std::cout << "--- error positions   ---" << std::endl;
  for (auto i = 0; i < k; i++) {
    if (decoded_bits[i] != info_bits[i]) {
      std::cout << i << ' ';
    }
  }
}
```

What could go wrong? Thank you in advance for any advice!

kouchy, 2020-10-23:
I fixed your code, the problem comes from a bad usage of the `info_bit_pos` vector. I will add a `throw` in the AFF3CT source code to check if this vector is valid ;-).

```cpp
// #include "codec.hpp"
#include "aff3ct.hpp"
#include "math.h"

std::string LDPC_H_MATRIX_NAME("MACKAY_504_1008.alist");

auto get_h_qc(std::string file_name) {
  std::ifstream h_file;
  h_file.open(file_name);
  auto h_mat = aff3ct::tools::QC::read(h_file);
  h_file.close();
  return h_mat;
}

auto get_h_alist(std::string file_name) {
  std::ifstream h_file (file_name);
  auto h_mat = aff3ct::tools::AList::read(h_file);
  h_file.close();
  return h_mat;
}

auto ldpc_encode(const int k, const int n, const std::vector<int> &info_bits,
	             std::vector<unsigned> &info_bits_pos) {
  // get h matrix
  auto h_mat = get_h_alist(LDPC_H_MATRIX_NAME);
  auto ldpc_encoder = aff3ct::module::Encoder_LDPC_from_H<int>(k, n, h_mat,
  	"IDENTITY", "", true, 1);

  // encode
  std::vector<int> encoded_bits(n);
  ldpc_encoder.encode(info_bits, encoded_bits);

  // copy the positions of the information bits in the `info_bits_pos` vector
  // the decoder will need this information to determine the `K` information
  // bits from the full codeword later
  std::copy(ldpc_encoder.get_info_bits_pos().begin(),
            ldpc_encoder.get_info_bits_pos().end(),
            info_bits_pos.begin());

  return encoded_bits;
}

auto ldpc_decode(const int k, const int n, const std::vector<float> &received,
                 const std::vector<unsigned> &info_bits_pos) {
  // generate h matrix
  auto h_mat = get_h_alist(LDPC_H_MATRIX_NAME);
  // std::vector<unsigned> read_info_bits_pos(k, 0);
  auto ldpc_decoder = aff3ct::module::Decoder_LDPC_BP_flooding_SPA<>(
      k, n, 30, h_mat, info_bits_pos);

  // decode
  std::vector<int> decoded_bits(k);
  ldpc_decoder.decode_siho(received, decoded_bits);

  return decoded_bits;
}

template <typename T> auto print_vector(std::vector<T> &vec) {
  std::for_each(vec.begin(), vec.end(),
                [&](const auto i) { std::cout << i << ' '; });
  std::cout << std::endl;
}

int main() {
  int k = 504;
  int n = 1008;
  float snr_max = 20;

  // generate information bits
  auto source =
      std::unique_ptr<aff3ct::module::Source_random<>>(new aff3ct::module::Source_random<>(k));
  std::vector<int> info_bits(k);
  source->generate(info_bits);

  // this is the position of the K information bits in the LDPC codeword, this 
  // is determined by the LDPC encoder
  std::vector<unsigned> info_bits_pos(k);

  // ldpc encode
  auto encoded_bits = ldpc_encode(k, n, info_bits, info_bits_pos);

  // generate gaussian noise
  const float mean = 0.0;
  const float stddev = std::sqrt(std::pow(10, -snr_max / 10.0));
  std::normal_distribution<float> dist(mean, stddev);
  std::default_random_engine generator;

  // awgn channel
  std::vector<float> received;
  std::transform(
      encoded_bits.begin(), encoded_bits.end(), std::back_inserter(received),
      [&](const int i) -> float { return -i * 2 + 1 + dist(generator); });

  // ldpc decode
  auto decoded_bits = ldpc_decode(k, n, received, info_bits_pos);

  // print vectors
  std::cout << "--- received symbols  ---" << std::endl;
  print_vector(received);
  std::cout << "--- info bits         ---" << std::endl;
  print_vector(info_bits);
  std::cout << "--- decoded bits      ---" << std::endl;
  print_vector(decoded_bits);

  // check for errors
  std::cout << "--- error positions   ---" << std::endl;
  for (auto i = 0; i < k; i++) {
    if (decoded_bits[i] != info_bits[i]) {
      std::cout << i << ' ';
    }
  }
  std::cout << std::endl;

  return 0;
}
```

kouchy, 2020-10-23:
I added the missing checks in the LDPC decoders to avoid the problem you encountered (see https://github.com/aff3ct/aff3ct/commit/7cee7d9b38e3e135c239769e0a664e34390f55f4).

SubtleMuffin, 2020-10-23:
@kouchy Thank you for the *best* answer!

closed 2020-10-24 by SubtleMuffin