aff3ct/aff3ct #88: unable to run aff3ct.exe

HindTh, 2020-10-28:
Hi all,

I am tired to open aff3ct.exe but its window appears for a few seconds and then closed.
This is the last step of the installation. 
[image]

kouchy, 2020-10-28:
Hi @HindTh,

Do you double click on `aff3ct.exe`? From the Windows graphic interface?

If it is the case, it is not the right usage of AFF3CT. AFF3CT is not a graphic application, it is a command line application and should be run from the command line.

I hope it helps.

HindTh, 2020-10-28:
> Hi @HindTh,
> 
> Do you double click on `aff3ct.exe`? From the Windows graphic interface?
> 
> If it is the case, it is not the right usage of AFF3CT. AFF3CT is not a graphic application, it is a command line application and should be run from the command line.
> 
> I hope it helps.

Dear @kouchy 

I am run the program from "MinGW Command Prompt" but the problem still exists.

HindTh, 2020-10-28:
I get this snapshot 
[image]

kouchy, 2020-10-28:
Good, AFF3CT is working now. You can try to run it with some parameters like this:
```
aff3ct.exe -C "POLAR" -K 1723 -N 2048 -m 1.0 -M 4.0 -s 1.0
```
Please read the [documentation](https://aff3ct.readthedocs.io/en/latest/) to more advanced usage.

closed 2020-10-28 by kouchy

HindTh, 2020-10-28:
> Good, AFF3CT is working now. You can try to run it with some parameters like this:
> 
> ```
> aff3ct-2.3.5.exe -C "POLAR" -K 1723 -N 2048 -m 1.0 -M 4.0 -s 1.0
> ```
> 
> Please read the [documentation](https://aff3ct.readthedocs.io/en/latest/) to more advanced usage.

Dear sir,

aff3ct may be working but it just flashing and close I cannot catch the window.