aff3ct/aff3ct #94: Is there a way to modify the Turbo code to increase the default ratio from 1/3 to e.g. 1/5 or 1/6, 1/7 etc ?

AtanasRusevPros, 2020-12-16:
Hello All,

I already use the framework but cannot find a valid way to do this. Can you give me any advises? I would like to encode with bigger Ratio.

Thank you,
Atanas

kouchy, 2020-12-18:
Hi @AtanasRusevPros,

There is no technique in AFF3CT to decrease the turbo code rate, to be honest I don't know if it is possible.
I would say that it is possible if you use new inner RSC codes with more than one parity bit, but you will have to build a new turbo code.

However, there is some techniques to increase the code rate (to have R = 1/2 or more for instance), see the documentation page here : https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/turbo/puncturer.html#pct-pattern).

I hope it helps.

AtanasRusevPros, 2020-12-18:
Hi Adrien,

I have made a mistake in my description - I wanted to Decrease the Ratio, i.e. making it e.g. 0.1 or 0.2, so that when decoding I have better performance with respect to the noise errors... currently it is ~0.3, meaning that 2016 bits are sent encoded as 6060. Up to 16% discrete bit error I get my data correct, but e.g. with 20% error (introduced by flipping each 5th bit) I get 41% error in my results.
And I want to try to encode 2016 bits with 1/5, resulting in message of roughly 10104 bits, which would be with stronger error resilience.....

So I guess simply Puncturer will not help me for this...

Any thoughts or options? Or this would mean that again I have to completely implement my own new Turbo encoder / decoder?

BR,
Atanas

kouchy, 2020-12-21:
Hi @AtanasRusevPros,

From my understanding of the convolutional codes: they are by construction limited to a fixed rate. Two convolutional codes of R=1/2 give a turbo code of rate R=1/3. If you seek for more flexibility in the code rate I think that the polar codes are more interesting... But you will have fixed codeword length by construction. World is never perfect :-).

I hope it helps.

AtanasRusevPros, 2020-12-21:
Hi @kouchy ,

thank you very much, any information definitely helps! I will try to see if I can add one more additional encoding stage, and further add it in the decoding calculations.

FYI - as far as I understood - the IT++ library offers some way of adding additional stages which lead to decreased R. But due to multiple reasons I'm not allowed to use it.

In any case I see that there is nothing more that can be said on the subject so I will close the issue!

Thank you and have a nice Christmas Holidays!

Atanas

closed 2020-12-21 by AtanasRusevPros