aff3ct/aff3ct #95: Decoder_SCF_naive

Aishusiva, 2020-12-17:
Could you please help me on how to use the function Decoder_SCF_naive function?I know it requires the parameters K,N,vector bool of frozen bits,n_flips.I am wondering how to declare and pass the other vector that the function expects.

kouchy, 2020-12-18:
Hi @Aishusiva,

The SC flip requires to be concatenated with a CRC (this is the only other parameter required in the constructor).
https://github.com/aff3ct/aff3ct/blob/development/include/Module/Decoder/Polar/SCF/Decoder_polar_SCF_naive.hpp#L33-L34

Aishusiva, 2021-01-19:
Does 'const CRC<B>& crc' denote the size of the CRC bits added to the input sequence?

Aishusiva, 2021-01-19:
Can CRC_polynomial be used with the SCF function?