aff3ct/aff3ct #98: Frozen_bit_generator_GA

Aishusiva, 2021-01-04:
I am a final year undergrad student. I am working with AFF3CT library. Could you please help me on how Frozen_bit_generator_GA function works? From my understanding of polar encoding technique, I thought the size of frozen bits should be N-K. But the function accepts only a vector of size N?

bonben, 2021-01-13:
GA implements Gaussian Approximation from trifonov. See https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/polar/encoder.html?highlight=Gaussian#enc-fb-gen-method.

The function accepts a vector of size N, in which 1s corresponds to frozen bits, and 0s to information bits.

Aishusiva, 2021-01-13:
Thanks a lot. Does the function produce the same set of frozen bit positions for all the SNRs?
I saw in the documentation that it is optimized for each SNR.I tried for different SNRs but the bits are the same. Could you please help me if I have gone wrong anywhere?

bonben, 2021-01-14:
Maybe you had a situation in which it did not change the frozen bits. It depends. If you put completely different SNR with large values of N you should observe differences.

Aishusiva, 2021-01-15:
Thank you so much @bonben

closed 2021-01-15 by Aishusiva