aff3ct/aff3ct #110: rate 1/2 in turbo encoder

frcgnn85, 2021-02-05:
Hi All,
There is a way to set the rate to 1/2 when using a turbo encoder? I think the simulator use by default the rate 1/3.
How the rate is calculated? 

Thanks a lot for your support.

Kind Regards,
Francesco

bonben, 2021-02-05:
Hi Francesco.

The rate of turbo codes is 1/3 by default.
But you can change the rate using a punturer : https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/turbo/puncturer.html?highlight=puncturer
For example in this [case from our comparator](http://aff3ct.github.io/comparator.html?curve0=e718fb7&xaxis=Eb%2FN0&yaxes=BER%2CFER) : 
`
    aff3ct --sim-type BFER -C TURBO -K 2048 -m 0.0 -M 2.0 -s 0.1 --enc-sub-std LTE -i 8 --dec-sf-type LTE_VEC --dec-sub-implem FAST --dec-sub-max MAX --pct-pattern "11,10,01" --crc-implem FAST --crc-poly 24-LTEA --dec-fnc --dec-fnc-ite-m 5
`
The rate is 1/2 thanks to the puncturer.

frcgnn85, 2021-02-05:
Hi Mathieu! Thanks a lot for your fast and esaustive reply! 

Could you please link me where the turbo decoder is implemented? In code I mean... 

I have a question also regarding the length of the encode bits (N)... why is four bits longer than the length indicate in the standard? I mean reading from the standard of I use an Information block length k equal to 3568, the Codeword length n should be 10716 with a r=1/3... but from simulation I got 10720... 4 bits more... 

Thanks a lot for your support!

Kind regards,
Francesco