aff3ct/aff3ct #111: SIM_THR (Mb/s) stats

frcgnn85, 2021-02-10:
Hi All,
the values in the column SIM_THR (Mb/s) are the decoded output throughput? I mean the bits per second exiting by the decoder?

Thanks a lot for your support!

Cheers,
Francesco

kouchy, 2021-02-10:
Hi @frcgnn85,

No this is the throughput of the whole simulation (see https://aff3ct.readthedocs.io/en/latest/user/simulation/overview/overview.html#output).
If you want the decoder throughput you should use the `--sim-stats` argument (see https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/simulation/simulation.html#sim-stats).

Hope it helps.

frcgnn85, 2021-02-10:
Hi Adrien, 
Thanks a lot for your reply!

I would like to analyze the turbo decoder for the ccsds standard... so I used the following command with the --sim--stats argument as you suggested.

./aff3ct --sim-type BFER -C TURBO -K 1784 -m 0.0 -M 1.0 -s 0.1 --enc-sub-std CCSDS -i 10 --dec-implem FAST --dec-sub-type BCJR --dec-sub-implem FAST --dec-sub-max MAX --sim-stats

For the last value of Eb/N0 (1.0) I got the following values in the  Measured throughput considering the last socket:

MODULE                    | AVERAGE (Mb/s) |  MINIMUM (Mb/s) |  MAXIMUM (Mb/s)
Decoder                      |        0.01               |           0.01             |            0.02
Channel                      |        4.74               |            1.29            |            6.12
Source                        |        20.94             |            1.59            |            31.88
Encoder                      |        73.88             |            4.61            |            129.31
Monitor                       |        31.02             |            1.44            |             44.70
Modem (modulate)     |        294.94           |            42.74          |             461.29
Modem (demodulate) |        367.99           |            6.74            |             654.71

So the maximum Decoder Throughput is 0.02 Mb/s, the minimum Decoder Throughput is 0.1Mb/s and the averaged one is 0.1 Mb/s? I'm right?

I also attached the all stats output (turbo_dec_stats
[turbo_dec_stats.txt](https://github.com/aff3ct/aff3ct/files/5958548/turbo_dec_stats.txt)
[turbo_dec_stats.txt](https://github.com/aff3ct/aff3ct/files/5958550/turbo_dec_stats.txt)

.txt) just in case.

Thanks a lot for any suggestions/help!

Cheers,
Francesco

kouchy, 2021-06-17:
Yes you're right, the CCSDS is based on a generic version of the BCJR decoder which is slow... Only BCJR decoder with LTE trellis has been optimized to run fast at this time.

closed 2021-06-17 by kouchy