aff3ct/aff3ct #125: Slower performance for Decoder_BCH_fast and Decoder_chase_pyndiah_fast

Divya916947, 2021-04-13:
I used Decoder_turbo_product with a. Decoder_BCH_std and Decoder_chase_pyndiah and
                                                         b. Decoder_BCH_fast and Decoder_chase_pyndiah_fast
But the decoding rate  of 'b' is lesser than 'a' implementation. Could you please tell me why?

OlivierHartmann, 2021-06-17:
Hi Divya916947, 
With which options did you compile aff3ct ? Did you specify the right architecture flag ?

Divya916947, 2021-06-17:
Hi OlivierHartmann 

I am using MS visual studio. I compiled the AFF3CT library with AVX2 option.

OlivierHartmann, 2021-06-17:
Please check that you used option like -O3 to compile. I guess you can check easily with a basic AWGN channel if you're installation is good. You have examples in [pyber](https://aff3ct.readthedocs.io/en/latest/user/simulation/pyber/pyber.html)

It makes years now, so I don't remember well, but SIMD are not the best with small codes. I can just tell you that BCH were really hard to enhance with SIMD as there are a lot of conditions in the algorithm, more if you use them in such a complex decoder that is the TPC. They may not be as fast as we hope.