aff3ct/aff3ct #126: please fix the website

Github-Shaw, 2021-05-18:
I can't download files from the website.help me
thank you very much

bonben, 2021-05-18:
Hi.

We just checked, it's working here http://aff3ct.github.io/download.html.
It must be a problem on your side.

BR.

Github-Shaw, 2021-06-02:
thanks, could you please help me solve another question?
i would like to use Frozenbits_generator to generate frozenbits, but it need an array best_channels, how could i get this?
i would like to use&nbsp; the class Encode_polar to encode the info, so i need to use the frozenbits_generator.
thank you very much.
i'm looking forward to your letter.

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "Mathieu ***@***.***&gt;; 
发送时间: 2021年5月18日(星期二) 下午3:16
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [aff3ct/aff3ct] please fix the website (#126)

嗨

我们刚检查过，它在这里工作http://aff3ct.github.io/download.html.
肯定是你那边出了问题。

比尔

—
您正在接收此消息，因为您创建了该线程。
直接回复此邮件，在 GitHub 上查看或取消订阅.

kouchy, 2021-06-17:
Please do not post different questions in the same thread.

closed 2021-06-17 by kouchy