aff3ct/aff3ct #134: Throughput of Decoder_BCH_fast lesser than Decoder_BCH_std

BonbenMathiew, 2021-06-03:
I tried implementing BCH encoding and decoding using AFF3CT library. I found that Decoder_BCH_fast runs slower than Decoder_BCH_std. Are there any compiler options that need to be enabled?
Any help would be greatly appreciated.
Thanks.

kouchy, 2021-06-17:
Hi @BonbenMathiew,

The `Decoder_BCH_fast` is not very efficient at this time, so if you are using it with many threads in parallel it does not surprise me that it can be slower than the `Decoder_BCH_std` :-(.

I hope it helps.

BonbenMathiew, 2021-06-18:
Hi Kouchy,

Thanks for the reply.

I am running the code on single thread. I find the performance of Decoder_BCH_fast is slower than Decoder_BCH_std. Is there a way I can speed up the performance?

Thank you.