aff3ct/aff3ct #146: Compile the static library with NDK

sbelleno-inria, 2021-09-08:
Hello,

We are using aff3ct as a third-party library inside our research cross-platform project.
When I tried to compile for Android using I got an error

> 'execinfo.h' file not found

How could I get rid of this error ? Thanks !

Here is the command line on the aff3ct CMakeLists.txt:
`
cmake -DCMAKE_TOOLCHAIN_FILE=/usr/lib/android-sdk/ndk/21.3.6528147/build/cmake/android.toolchain.cmake -DANDROID_NATIVE_API_LEVEL=29 -DANDROID_TOOLCHAIN=clang -DANDROID_STL=c++_shared -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DAFF3CT_COMPILE_EXE:BOOL=FALSE -DAFF3CT_COMPILE_STATIC_LIB:BOOL=TRUE -DAFF3CT_POLAR_BIT_PACKING:BOOL=FALSE -DAFF3CT_COLORS:BOOL=FALSE
`

Here is the full trace:

```console
-- AFF3CT - Version: 2.3.5
-- AFF3CT - Compile: static library
-- AFF3CT - System: Unix/Linux
-- AFF3CT - Polar bit packing: off
-- AFF3CT - Polar bounds: off
-- AFF3CT - Terminal colors: off
-- AFF3CT - Backtrace: on
-- AFF3CT - External strings: on
-- AFF3CT - Precision: 8/16/32/64-bit
-- AFF3CT - Header found: MIPP
-- AFF3CT - Header found: rang
-- AFF3CT - Header found: date
-- Configuring done
-- Generating done
-- Build files have been written to: /builds/icov-demonstrator/build/release/android/core/third-party/aff3ct
[  0%] Building CXX object CMakeFiles/aff3ct-obj.dir/src/Factory/Launcher/Launcher.cpp.o
[  1%] Building CXX object CMakeFiles/aff3ct-obj.dir/src/Factory/Module/CRC/CRC.cpp.o
[  2%] Building CXX object CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Codec/Codec_SIHO_HIHO.cpp.o
[  2%] Building CXX object CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Codec/Codec_HIHO.cpp.o
[  2%] Building CXX object CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Codec/BCH/Codec_BCH.cpp.o
[  2%] Building CXX object CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Channel/Channel.cpp.o
[  3%] Building CXX object CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Codec/Codec.cpp.o
[  2%] Building CXX object CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Codec/Codec_SIHO.cpp.o
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Factory/Module/Codec/Codec_SIHO_HIHO.cpp:1:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Factory/Module/Codec/BCH/Codec_BCH.hpp:8:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/CRC/CRC.hpp:15:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Module.hpp:18:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Task.hpp:10:
/builds/icov-demonstrator/core/third-party/aff3ct/lib/MIPP/src/mipp.h:90:10: fatal error: 'execinfo.h' file not found
#include <execinfo.h>
         ^~~~~~~~~~~~
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Factory/Module/Codec/Codec_SIHO.cpp:1:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Factory/Module/Codec/BCH/Codec_BCH.hpp:8:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/CRC/CRC.hpp:15:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Module.hpp:18:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Task.hpp:10:
/builds/icov-demonstrator/core/third-party/aff3ct/lib/MIPP/src/mipp.h:90:10: fatal error: 'execinfo.h' file not found
#include <execinfo.h>
         ^~~~~~~~~~~~
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Factory/Module/Codec/BCH/Codec_BCH.cpp:1:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Factory/Module/Encoder/BCH/Encoder_BCH.hpp:9:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Encoder/BCH/Encoder_BCH.hpp:7:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Encoder/Encoder.hpp:15:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Module.hpp:18:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Task.hpp:10:
/builds/icov-demonstrator/core/third-party/aff3ct/lib/MIPP/src/mipp.h:90:10: fatal error: 'execinfo.h' file not found
#include <execinfo.h>
         ^~~~~~~~~~~~
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Factory/Module/Codec/Codec_HIHO.cpp:1:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Factory/Module/Codec/BCH/Codec_BCH.hpp:8:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/CRC/CRC.hpp:15:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Module.hpp:18:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Task.hpp:10:
[... 42 lines cut ...]
CMakeFiles/aff3ct-obj.dir/build.make:172: recipe for target 'CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Codec/Codec_HIHO.cpp.o' failed
make[2]: *** [CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Codec/Codec_HIHO.cpp.o] Error 1
1 error generated.
CMakeFiles/aff3ct-obj.dir/build.make:185: recipe for target 'CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Codec/Codec_SIHO.cpp.o' failed
make[2]: *** [CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Codec/Codec_SIHO.cpp.o] Error 1
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Factory/Launcher/Launcher.cpp:14:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Launcher/Launcher.hpp:20:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Simulation/Simulation.hpp:19:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Module.hpp:18:
In file included from /builds/icov-demonstrator/core/third-party/aff3ct/src/Module/Task.hpp:10:
/builds/icov-demonstrator/core/third-party/aff3ct/lib/MIPP/src/mipp.h:90:10: fatal error: 'execinfo.h' file not found
#include <execinfo.h>
         ^~~~~~~~~~~~
1 error generated.
CMakeFiles/aff3ct-obj.dir/build.make:133: recipe for target 'CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Channel/Channel.cpp.o' failed
make[2]: *** [CMakeFiles/aff3ct-obj.dir/src/Factory/Module/Channel/Channel.cpp.o] Error 1
1 error generated.
CMakeFiles/aff3ct-obj.dir/build.make:107: recipe for target 'CMakeFiles/aff3ct-obj.dir/src/Factory/Launcher/Launcher.cpp.o' failed
make[2]: *** [CMakeFiles/aff3ct-obj.dir/src/Factory/Launcher/Launcher.cpp.o] Error 1
CMakeFiles/Makefile2:123: recipe for target 'CMakeFiles/aff3ct-obj.dir/all' failed
make[1]: *** [CMakeFiles/aff3ct-obj.dir/all] Error 2
Makefile:170: recipe for target 'all' failed
make: *** [all] Error 2

```

kouchy, 2021-09-10:
Hi @sbelleno-inria,

The problem come from system functions that are not available (or not the same) on Android. These functions are mainly used to get the backtrace when AFF3CT crashes. I disabled this feature on Android on the commit https://github.com/aff3ct/aff3ct/commit/5326b0db4f573f753f5fd5fcaef10ce11a1ee653 (`development` branch) by using the `__ANDROID__` macro.

Could you check if it compiles now?

sbelleno-inria, 2021-09-10:
Hello,

thank you for your response, now the compilation is working.

closed 2021-09-10 by sbelleno-inria