aff3ct/aff3ct #150: Compile as na library on Windows

SubtleMuffin, 2021-11-04:
Hey @kouchy , hope you are doing great.

I was trying to compile `aff3ct` as a library and work with the example `cython_polar` (in dev branch of `my_project_with_aff3ct`). I discovered couple of issues while trying to do so:

1. The compiler `cl.exe` was complaining about `mipp_scalar_op.hxx`, but comment out the lines
```c++
template <          > inline int16_t add(const int16_t val1, const int16_t val2) { return (int16_t)std::min(std::max((int32_t)((int32_t)val1 + (int32_t)val2),(int32_t)std::numeric_limits<int16_t>::min()),(int32_t)std::numeric_limits<int16_t>::max()); }
template <          > inline int8_t  add(const int8_t  val1, const int8_t  val2) { return (int8_t )std::min(std::max((int16_t)((int16_t)val1 + (int16_t)val2),(int16_t)std::numeric_limits<int8_t >::min()),(int16_t)std::numeric_limits<int8_t >::max()); }
```
does the trick.

Here is the log
```
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ilib/include -Ilib/lib/cli/src -Ilib/lib/MIPP/src -Ilib/lib/rang/include -Ilib -IC:\Users\username\Miniconda3\envs\research\include -IC:\Users\username\Miniconda3\envs\research\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\include -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /EHsc /Tpcodec.cpp /Fobuild\temp.win-amd64-3.9\Release\codec.obj
codec.cpp
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(12): warning C4003: not enough arguments for function-like macro invocation 'min'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(12): warning C4003: not enough arguments for function-like macro invocation 'max'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(12): error C2589: '(': illegal token on right side of '::'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(12): error C2062: type 'unknown-type' unexpected
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(12): error C2059: syntax error: ')'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(13): warning C4003: not enough arguments for function-like macro invocation 'min'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(13): warning C4003: not enough arguments for function-like macro invocation 'max'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(13): error C2589: '(': illegal token on right side of '::'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(13): error C2062: type 'unknown-type' unexpected
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(13): error C2059: syntax error: ')'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(16): warning C4003: not enough arguments for function-like macro invocation 'min'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(16): warning C4003: not enough arguments for function-like macro invocation 'max'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(16): error C2589: '(': illegal token on right side of '::'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(16): error C2062: type 'unknown-type' unexpected
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(16): error C2059: syntax error: ')'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(17): warning C4003: not enough arguments for function-like macro invocation 'min'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(17): warning C4003: not enough arguments for function-like macro invocation 'max'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(17): error C2589: '(': illegal token on right side of '::'
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(17): error C2062: type 'unknown-type' unexpected
C:\Users\username\aff3ct\lib\lib\MIPP\src\mipp_scalar_op.hxx(17): error C2059: syntax error: ')'
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30037\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
```

2. Now, in the Python's `setup.py` file,
```python
codec_polar = Extension(
    name="codec_polar",
    sources=["codec_polar.pyx"],
    libraries=[library],
    library_dirs=["../../lib/aff3ct/build/lib"],
    include_dirs=["../../lib/aff3ct/include", "../../lib/aff3ct/lib/cli/src", "../../lib/aff3ct/lib/MIPP/src", "../../lib/aff3ct/lib/MIPP/src", "../../lib/aff3ct/lib/rang/include"],
    language="c++",
    extra_compile_args=["-std=c++11"],
    extra_link_args=["-std=c++11"]
)
```
It requires `libraries` and `library_dirs`. I tried the `/build/lib/Release` and `/build/lib`, which contains the following folders `cli`, `cmake` and `Release`. The last folder contains `aff3ct-3.0.1.dll` and `aff3ct-3.0.1.lib`. But the Python still tells me these

```
warning: no library file corresponding to 'lib/build/lib/Release/aff3ct-3.0.1.lib' found (skipping)
```
and consequently, `link.exe` tells me
```
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:lib/build/lib /LIBPATH:C:\Users\username\Miniconda3\envs\research\libs /LIBPATH:C:\Users\username\Miniconda3\envs\research\PCbuild\amd64 /LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\lib\x64 /LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x64 /LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\um\x64 /EXPORT:PyInit_codec build\temp.win-amd64-3.9\Release\codec.obj /OUT:C:\Users\username\aff3ct\codec.cp39-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.9\Release\codec.cp39-win_amd64.lib
   Creating library build\temp.win-amd64-3.9\Release\codec.cp39-win_amd64.lib and object build\temp.win-amd64-3.9\Release\codec.cp39-win_amd64.exp
codec.obj : error LNK2001: unresolved external symbol "public: static class aff3ct::tools::Sparse_matrix __cdecl aff3ct::tools::QC::read(class std::basic_istream<char,struct std::char_traits<char> > &)" (?read@QC@tools@aff3ct@@SA?AVSparse_matrix@23@AEAV?$basic_istream@DU?$char_traits@D@std@@@std@@@Z)
codec.obj : error LNK2001: unresolved external symbol "public: __cdecl aff3ct::tools::Frozenbits_generator_GA_Arikan::Frozenbits_generator_GA_Arikan(int,int,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,bool)" (??0Frozenbits_generator_GA_Arikan@tools@aff3ct@@QEAA@HHAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@_N@Z)
codec.obj : error LNK2001: unresolved external symbol "public: __cdecl aff3ct::tools::invalid_argument::invalid_argument(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (??0invalid_argument@tools@aff3ct@@QEAA@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@H00@Z)
codec.obj : error LNK2001: unresolved external symbol "public: __cdecl aff3ct::tools::length_error::length_error(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (??0length_error@tools@aff3ct@@QEAA@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@H00@Z)
codec.obj : error LNK2001: unresolved external symbol "public: __cdecl aff3ct::tools::unimplemented_error::unimplemented_error(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,int,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (??0unimplemented_error@tools@aff3ct@@QEAA@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@H00@Z)
[... cut ...]
codec.obj : error LNK2001: unresolved external symbol "public: __cdecl aff3ct::module::Encoder_polar<int>::Encoder_polar<int>(int const &,int const &,class std::vector<bool,class std::allocator<bool> > const &)" (??0?$Encoder_polar@H@module@aff3ct@@QEAA@AEBH0AEBV?$vector@_NV?$allocator@_N@std@@@std@@@Z)
C:\Users\username\aff3ct\codec.cp39-win_amd64.pyd : fatal error LNK1120: 63 unresolved externals
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30037\\bin\\HostX86\\x64\\link.exe' failed with exit code 1120
```

To me, it looks like the linker cannot find the libraries... could you help me with this?

kouchy, 2021-11-05:
Hi @SubtleMuffin,

For the compilation of MIPP `add` functions: I guess you included `windows.h` somewhere and you are not defining `#define NOMINMAX` (see: https://stackoverflow.com/questions/13416418/define-nominmax-using-stdmin-max). Maybe you should try to add `-DNOMINMAX` when you compile.

Just a remark: if you are using `mipp_scalar_op.hxx` it means that the MIPP vectorization is disabled. As a consequence, the AFF3CT code will be slower. You can try these flags for a faster code: `-DNOMINMAX -D_CRT_SECURE_NO_DEPRECATE /MT /EHsc /arch:AVX2   -D__SSE__ -D__SSE2__ -D__SSE3__ -D__SSSE3__ -D__SSE4_1__ -D__SSE4_2__ -D__AVX__ -D__AVX2__ -D__FMA__`

For your second problem, honestly I don't know :-/. However it works in the tests. Here `cython_polar` has been successfully built: https://gitlab.com/aff3ct/my_project_with_aff3ct/-/jobs/1653319527. I suggest you to do the same compilation steps as here: https://github.com/aff3ct/my_project_with_aff3ct/blob/development/.gitlab-ci.yml#L148-L172.

Also I don't know if you saw but there is some work on a different approach to wrap AFF3CT in Python. The wrapping approach is automated and many AFF3CT modules are working now (including the polar codes). Here is the repository of the project: https://github.com/aff3ct/py_aff3ct/tree/master. And here you can see an example of how to use polar codes: https://github.com/aff3ct/py_aff3ct/blob/master/examples/full_python/bfer_polar.py.
If it is hard for you to understand the philosophy, there is some detailed notebooks, the simplest one is: https://github.com/aff3ct/py_aff3ct/blob/master/examples/notebooks/hello_world.ipynb.

`py_aff3ct` is able to run very fast (very close to the C code) compared to `cython_polar`.

I hope it helps.

SubtleMuffin, 2021-11-22:
@kouchy Thank you so much for your reply! I am so sorry for late reply.

I have solved the issue, and I will bring a new PR on the example repo. The root cause is the Python script (`setup.py`) in that example gives wrong answers about paths on Windows.

The `py_aff3ct` project is really awesome! But so far I will need to run the compiled binaries (you know, shared lib + Python wrapper) on Windows due to device driver constraints... I will keep a close eye on the progress of `py_aff3ct`!

closed 2021-11-22 by SubtleMuffin