aff3ct/aff3ct #153: How to work with my own LDPC code sample

ElifUskuplu, 2022-02-28:
Hi, I want to use AFF3CT to simulate my codes. I have the folllowing matrix written in Matlab. How can I convert this to a suitable data for the application?

````

k=43;

DCA=[];
Blocks=[];

%DCA(3,2n+1,2n+1) 

for g=1:3
    for j=1:(k)
        if g==1
           DCA(j,g)=0;
        elseif g==2
           DCA(j,g)=j-1;
        else
           DCA(j,g)= mod(4.*(j-1),k);    
        end
    end
end

disp(DCA);

%Blocks and Incidence matrix
for j=1:k
        for a=1:k
               Bj(a,:)=[DCA(j,2) a-1+k mod(DCA(j,2)+a-1,k)+2*k mod(DCA(j,3)+a-1,k)+3*k];
        end
        Blocks=[Blocks; Bj];
        
        for a=1:k
          for i=1:4*k
              if isempty(find(Bj(a,:)==i-1))==false
                  H(i,a+(j-1)*k)=1;
              else
                  H(i,a+(j-1)*k)=0;
              end
          end
        end
end
%disp(Blocks);

end

````

kouchy, 2022-03-03:
Hi @ElifUskuplu,

AFF3CT is able to read H partity matrices from AList or QC format. The expected formats are described here in the documentation : https://aff3ct.readthedocs.io/en/latest/user/simulation/parameters/codec/ldpc/decoder.html#dec-h-path-image-required-argument.

So the work you have to do is to print your H matrix in a text file, once it is done you will able to configure AFF3CT with the generated input H matrix with the `--dec-h-path` command line parameter.

I hope it helps.

ElifUskuplu, 2022-03-06:
Thanks for the help. 

I converted my sample to an alist file. I simulated the following codes, but I think there is a problem with the second one "ldpc43". Although, the small (20x25) ldpc code gave good results, the longer one didn't work properly.  I attached the alist formatted matrices and their results. How can I simulate the other ldpc matrix (172x1849)?

[ldpc5.txt](https://github.com/aff3ct/aff3ct/files/8191737/ldpc5.txt)

[result5.txt](https://github.com/aff3ct/aff3ct/files/8191742/result5.txt)

[ldpc43.txt](https://github.com/aff3ct/aff3ct/files/8191735/ldpc43.txt)

[result43.txt](https://github.com/aff3ct/aff3ct/files/8191743/result43.txt)

closed 2022-03-14 by ElifUskuplu

ibarron9216, 2022-06-01:
Hi @kouchy,

I wanted to ask a question regarding LDPC codes that comply with the 5G NR standard, which is mentioned in your paper that AFF3CT does support. However, instead of opening a new issue, I think this is a good place to ask, since the question is related to parity check matrix.

To use 5G NR LDPC codes, I need to provide the sparse parity check matrix, which is part of the aforementioned standard, right? specifically, I indicate with --dec-h-path the location of the matrix in .alist format.

I am using AFF3CT 2.3.5, and based on the names of the matrix in .alist format, I think the matrix for the 5G NR standard are not included in the conf/dec/LDPC folder; I think this makes sense because my understanding is that 5G NR LDPC codes encode the input data according to the parity-check matrix associated with the corresponding lifting size applied to the selected base graph from the 5G standard.

I hope that you can let me know if I am missing something. Last but not least, thank you for the amazing work you have done!