aff3ct/aff3ct #157: The aff3ct lib version of "my_project_with_aff3ct" is v2.3.5?

Hashiao, 2022-04-21:
I cloned aff3ct v3.0.1 and compiled the project to *lib in debug mode. But when I tried to call aff3ct library as the first example "bootstrap" shows, the IDE told  me that XXX has no member YYY, which means the called interface doesn't exist in aff3ct v3.0.1. So I checked the last commit date of "my_project_with_aff3ct". It seems that the aff3ct version of  "my_project_with_aff3ct" and the examples in formal documents is v2.3.5. I clone aff3ct v2.3.5 and verified my guess. I wonder why nobody comes up with this issue?

kouchy, 2022-04-21:
Hi @Hashiao,

I'm not sure to understand your problem.
Did you clone the `development` or the `master` branch of `my_project_with_aff3ct` (which commit)? I recommend to clone the `development` branch.
I checked the `CMakeLists.txt` file (https://github.com/aff3ct/my_project_with_aff3ct/blob/development/examples/bootstrap/CMakeLists.txt#L23) and indeed we forgot to replace `2.3.2` by `3.0.1`. Maybe it is your problem?

I hope it helps.

Hashiao, 2022-04-21:
> Hi @Hashiao,
>
> I'm not sure to understand your problem. Did you clone the `development` or the `master` branch of `my_project_with_aff3ct` (which commit)? I recommend to clone the `development` [...]

Thank you so much, now I can call aff3ct functions successfully. Your project helps so much.

closed 2022-04-21 by Hashiao