aff3ct/aff3ct #158: i want to use tools::Codec_LDPC object in my own project

Hashiao, 2022-05-09:
Hi, thanks again for your patience.
The version is aff3ct 3.0.1.
I want to test the performance of LDPC code, and I can use tools::Codec_LDPC when the --enc-type is LDPC_DVBS2. It seems all parameters (like K, N, Hmat)are stipulated in DVBS2_constants_16200.hpp/DVBS2_constants_64800.hpp. 
However, when I tried to use tools::Codec_LDPC when the --enc-type is LDPC_H, an error "vector out of range" is thrown. The error occurs when the module codec is initializing (m.codec = std::uniqe_ptr<tools::Codec_LDPC<>>(p.codec->build())). After a few functions called, the out of range occurs at LDPC_matrix_handler.cpp line 506, where the code wants to fill vector G[i][i-M] with 1, but the position is not memory allocated.

Here is my code, the H mat file is given by aff3ct source code.

// main.cpp

'''

#include <iostream>
#include <string>
#include <vector>

#include <aff3ct.hpp>
using namespace aff3ct;
#include "structs.h"
#include "func.h"

int main() {
	int K = 1664;
	int N = 2048;
	std::string matHPath = "D:/aff3ct-235/aff3ct-ldpc-demo/aff3ctLDPC/aff3ctLDPC/conf/dec/LDPC/10GBPS-ETHERNET_1723_2048.alist";
	// "D:/aff3ct-235/aff3ct-ldpc-demo/aff3ctLDPC/aff3ctLDPC/conf/dec/LDPC/MACKAY_504_1008.alist"
	// "D:/aff3ct-235/aff3ct-ldpc-demo/aff3ctLDPC/aff3ctLDPC/conf/dec/LDPC/WIMAX_288_576.alist"
	// "D:/aff3ct-235/aff3ct-ldpc-demo/aff3ctLDPC/aff3ctLDPC/conf/dec/LDPC/DEBUG_6_3.alist";

	Params p;
	Modules m;
	Utils u;
	Cmdargs cmdargs;
        // transfer parameters to cmd arguments
	init_cmdargs(K, N, matHPath, cmdargs);

	std::cout << "==================\nprint cmd args" << std::endl;
	for (int iarg = 0; iarg < cmdargs.argc; iarg++) {
		std::cout << "argv[" << iarg << "] = " << cmdargs.argv[iarg] << std::endl;
	}
	std::cout << "print cmd args done\n==================" << std::endl;

	init_params(cmdargs.argc, cmdargs.argv, p);
	init_modules(p, m);
	init_utils(p, m, u);

	// display the legend in the terminal
	u.terminal->legend();

	// sockets binding (connect the sockets of the tasks = fill the input sockets with the output sockets)
	using namespace module;
	(*m.encoder)[enc::sck::encode::U_K].bind((*m.source)[src::sck::generate::U_K]);
	(*m.modem)[mdm::sck::modulate::X_N1].bind((*m.encoder)[enc::sck::encode::X_N]);
	(*m.channel)[chn::sck::add_noise::X_N].bind((*m.modem)[mdm::sck::modulate::X_N2]);
	(*m.modem)[mdm::sck::demodulate::Y_N1].bind((*m.channel)[chn::sck::add_noise::Y_N]);
	(*m.decoder)[dec::sck::decode_siho::Y_N].bind((*m.modem)[mdm::sck::demodulate::Y_N2]);
	(*m.monitor)[mnt::sck::check_errors::U].bind((*m.encoder)[enc::sck::encode::U_K]);
	(*m.monitor)[mnt::sck::check_errors::V].bind((*m.decoder)[dec::sck::decode_siho::V_K]);

	std::vector<float> sigma(1);
	(*m.channel)[chn::sck::add_noise::CP].bind(sigma);
	(*m.modem)[mdm::sck::demodulate::CP].bind(sigma);

	// loop over the various SNRs
	for (auto ebn0 = p.ebn0_min; ebn0 < p.ebn0_max; ebn0 += p.ebn0_step)
	{
		// compute the current sigma for the channel noise
		const auto esn0 = tools::ebn0_to_esn0(ebn0, p.R, p.modem->bps);
		std::fill(sigma.begin(), sigma.end(), tools::esn0_to_sigma(esn0, p.modem->cpm_upf));
		u.noise->set_values(sigma[0], ebn0, esn0);

[... 190 lines cut ...]
	// create a sigma noise type
	u.noise = std::unique_ptr<tools::Sigma<>>(new tools::Sigma<>());
	// report the noise values (Es/N0 and Eb/N0)
	u.reporters.push_back(std::unique_ptr<tools::Reporter>(new tools::Reporter_noise<>(*u.noise)));
	// report the bit/frame error rates
	u.reporters.push_back(std::unique_ptr<tools::Reporter>(new tools::Reporter_BFER<>(*m.monitor)));
	// report the simulation throughputs
	u.reporters.push_back(std::unique_ptr<tools::Reporter>(new tools::Reporter_throughput<>(*m.monitor)));
	// create a terminal that will display the collected data from the reporters
	u.terminal = std::unique_ptr<tools::Terminal>(p.terminal->build(u.reporters));
}
'''

Besides, when I tried to called aff3ct-3.0.1.exe, there are some issues confused me. Take the h mat file "10GBPS-ETHERNET_1723_2048.alist" as example, from the file name, the K and N can be deduced, which are 1723 and 2048 correspondingly. However, when the aff3ct-3.0.1.exe is called by the following cmd args:

aff3ct-3.0.1.exe -C LDPC --enc_type LDPC_H --dec-h-path **10GBPS-ETHERNET_1723_2048.alist -K 1723 -N 2048 -m 0.0 -M 3.0 -s 1.0

a red (EE) message will show: In the ***/Encoder_LDPC.cpp file at line 77 ('_check-G_dimensions' function): "The built G matrix has a dimension 'K' different than the given one ('K' = 1723, 'G.get_n_cols()' = 1664)"
It can be deduced  from the red (EE) message that K should be 1664 but not 1723. So after I modified the args, the simulation successfully started:

aff3ct-3.0.1.exe -C LDPC --enc_type LDPC_H --dec-h-path **10GBPS-ETHERNET_1664_2048.alist -K 1723 -N 2048 -m 0.0 -M 3.0 -s 1.0

# End of the simulation.

So, I changed the K from 1723 to 1664 in my own code, but the error still occurs when a G mat is calculated from the given H mat(vector out of range). Do i need a module::Puncturer? Or I missed some unique procedures/parameters to initialize LDPC encoder/decoder?

Hashiao, 2022-05-11:
Hi every one!
I successfully called LDPC simulation where --enc-type == LDPC_H
I am not very familiar with the construction and theory of LDPC, so I just tried several parameters and some of them can work!

Two places need to be modified:
1 funcs.cpp function void init_cmdargs(int K, ......)

`

void init_cmdargs(int K, int N, std::string &matHPath, Cmdargs& cmdargs) {
	cmdargs.argc = 11;
	// init memory for argv[][]
	cmdargs.argv = new char* [cmdargs.argc];
	for (int iarg = 0; iarg < cmdargs.argc; iarg++) {
		cmdargs.argv[iarg] = new char[255];
	}
	// fill cmd args
	cmdargs.argv[0] = (char*)"*.exe";
	cmdargs.argv[1] = (char*)"-K";
	sprintf(cmdargs.argv[2], "%d", K);   

	cmdargs.argv[3] = (char*)"-N";
	sprintf(cmdargs.argv[4], "%d", N);   

	cmdargs.argv[5] = (char*)"--enc-type";
	cmdargs.argv[6] = (char*)"LDPC_H";   // 

	cmdargs.argv[7] = (char*)"--dec-h-path";
	cmdargs.argv[8] = (char*)matHPath.data();
        // in this way, only regular matrices are supported
        cmdargs.argv[9] = (char*)"--enc-g-method";
	cmdargs.argv[10] = (char*)"LU_DEC";
}

`

2 main.cpp   function int main()

`

int main() {
        int K = 288;
	int N = 576;
	std::string matHPath = "D:/aff3ct-235/aff3ct-ldpc-demo/aff3ctLDPC/aff3ctLDPC/conf/dec/LDPC/WIMAX_288_576.alist";

        // the following codes are kept the same
}

`

I am learning the theory of LDPC. If anyone wants to discuss LDPC coding, you may contact me via Email 894149540@qq.com or reply this issue.

kouchy, 2022-05-11:
Good to hear!

closed 2022-07-22 by kouchy