aff3ct/aff3ct #164: Reed-Solomon code in GENIUS decoding mode seems not working.

furkan-ercan, 2023-01-12:
Command:
 ./aff3ct-3.0.2 -C "RS" -K 239 -N 255 -m 7.75 -M 9.0 -s 0.25 -E ESN0 --dec-implem GENIUS
Output:
# The simulation is running...
(EE) Module clone failed when trying to duplicate the sequence: module name is 'Decoder_RS_genius'.
(EE) std::exception
# End of the simulation.

kouchy, 2023-01-15:
Hi @furkan-ercan,

The error message `(EE) Module clone failed when trying to duplicate the sequence: module name is 'Decoder_RS_genius'.` tells that the decoder RS genius can't be duplicated. It means that it cannot be run in parallel.
You may try to run it on 1 thread as follow (add `-t 1` and the end of the cmd line):
```
./aff3ct-3.0.2 -C "RS" -K 239 -N 255 -m 7.75 -M 9.0 -s 0.25 -E ESN0 --dec-implem GENIUS -t 1
```

I hope it helps.

furkan-ercan, 2023-01-16:
Hi Adrien,

Thanks a lot for the detailed clarification. All the best.

Furkan

closed 2023-01-16 by kouchy