aff3ct/aff3ct #171: Building with GCC13

SubtleMuffin, 2023-06-13:
Hey @kouchy -- I recently tried to build the library with GCC13, but got a few errors due to the compiler did not know what is a `uint8_t`. This breaking change is introduced by GCC 13, and please see more details [here](https://gcc.gnu.org/gcc-13/porting_to.html).

A silly but nonetheless working workaround is to include `cstdint` in a funky file `src/Tools/Perf/Transpose/transpose_selector.cpp` and it *should* build successfully. Another potential location is `main.cpp`.

Could we improve the library by explicitly including those libraries in a better place? Thanks!

kouchy, 2023-06-14:
Did you try to compile with the last development version? Is is still not working?

SubtleMuffin, 2024-01-04:
Thanks @gerhardbaer! Now the issue can be closed.

PR #181 (2023-12-16): transpose_selector: add missing includes
Add missing includes (<cstdint>) to fix compilation with GCC 13.

Closes #171
commit ac158349 (2023-12-16): transpose_selector: add missing includes

Add missing includes (<cstdint>) to fix compilation with GCC 13.

Closes #171
closed 2024-01-04 by SubtleMuffin