aff3ct/aff3ct #182: POLAR AVX2 tests fail

gerhardbaer, 2023-12-15:
Hi,

When the current `development` branch is compiled with `-march=x86-64-v3` (thus enabling AVX2), two tests fail in the POLAR suite.

For instance, the following would fail:
`aff3ct -p "8" --sim-type "BFER" -C "POLAR" -K "27307" -N "32768" -m "1.0" -M "4.0" -e "500" --enc-fb-gen-method "FILE" --enc-fb-awgn-path "conf/cde/awgn_polar_codes/TV/15/N32768_awgn_s0.489.pc" --dec-type "SC" --src-type "RAND" --src-implem "FAST" --chn-type "AWGN" --chn-implem "FAST" --dec-implem "FAST" --dec-simd "INTER" --qnt-bits "6" --qnt-dec "1" --qnt-implem "FAST" -e "100" --sim-stop-time "5" -R "1.0,1.1,1.2,1.3,1.4,1.5,1.6,1.7,1.8,1.9,2.0,2.1,2.2,2.3,2.4,2.5,2.6,2.7,2.8,2.9,3.0,3.1,3.2,3.3,3.4,3.5,3.6,3.7,3.8,3.9,4.0" -E "EBN0" 
`
Reported error:
transpose_selector.cpp' file at line 33 ('char_transpose' function): "'src' is unaligned memory."

This is because `Decoder_SIHO::Y_N` is a plain `std::vector<R>` without proper alignment, thus causing `Decoder_polar_SC_fast_sys<>::_load()` to pass unaligned data to `char_transpose()`.

As a side note: `_decode_siho()` did not complain because `API_polar::isAligned(Y_N)` return true if `MIPP_ALIGNED_LOADS` is not defined.

I can imagine of two (non disjoint) ways to solve this issue:
1. Make `aff3ct::tools::char_transpose()` fallback gracefully to SSE4.1 code path if alignment for AVX2 code path is not met ;
2. Make toplevel `Decoder_SIHO::Y_N` aligned "decently". The mipp::allocator could be used, but I'd rather see more improvements in there too. For one, on x86_64 systems, it's commonly admitted that 64B alignment (a cache line) is good for all SIMD sizes. Next, for "large enough" std::vector<> size (heuristics to be found), it could be interesting to align on page size.

kouchy, 2023-12-16:
Hi Gerhard,

Thank you for this extensive report and for your involvement in AFF3CT.

I see the problem. Before, all the buffers were correctly aligned in the `runtime::Task` class thx to MIPP aligned allocator.
But, recently we split AFF3CT into AFF3CT+AFF3CT-core. We decided to do not add MIPP as a dependency in AFF3CT-core as it was used ONLY for data allocation in the `runtime::Task` class.

So, I think that an elegant solution would be to re-align allocated data in the `runtime::Task` but without using MIPP.
All the data buffers used in AFF3CT are allocated here: 
  - https://github.com/aff3ct/aff3ct-core/blob/development/include/Runtime/Task/Task.hpp#L70

I think there is no problem to use a dedicated allocator in these `std::vector`s. Maybe a quick and dirty solution could be to align data buffers on 512 bits ? This way it should work with most of the current architectures...

gerhardbaer, 2023-12-16:
Agreed, aligning onto 64-byte boundaries sounds a good approach, especially on x86 (size of a cache line). Should be fine for arm architectures too, though typical cache line is 16 to 32 bytes IIRC. This remains a multiple of a CL and associativity should be effective.

Still, I would suggest the linked patch for (1). We probably should still use a custom allocator for `Decoder_SIHO` (2), as  `Y_N` is bound to the socket, isn't it?

gerhardbaer, 2023-12-22:
New PR against aff3ct-core now available.

closed 2023-12-23 by kouchy