aff3ct/aff3ct #184: How to use Reed-Solomon encoder/decoder

LiangJunqiang, 2023-12-20:
I'm trying to work with the aff3ct Reed-Solomon codec, and here's the code I modified:
``` c++
#include <iostream>
#include <memory>
#include <vector>
#include <string>

#include <aff3ct.hpp>
#include "cxxopts.hpp"

using namespace aff3ct;

struct params
{
	int   K         ; // number of information bits
	int   N         ; // codeword size
	int   fe        ; // number of frame errors
	int   seed      ; // PRNG seed for the AWGN channel
	float ebn0_min  ; // minimum SNR value
	float ebn0_max  ; // maximum SNR value
	float ebn0_step ; // SNR step
	float R         ; // code rate (R=K/N)
};

struct modules
{
	std::unique_ptr<module::Source_random<>>          source;
	std::unique_ptr<module::Encoder_RS<>> 		      encoder;
	std::unique_ptr<module::Modem_BPSK<>>             modem;
	std::unique_ptr<module::Channel_AWGN_LLR<>>       channel;
	std::unique_ptr<module::Decoder_RS_std<>>         decoder;
	std::unique_ptr<module::Monitor_BFER<>>           monitor;
};

struct buffers
{
	std::vector<int  > ref_bits;
	std::vector<bool > frozen_bits;
	std::vector<int  > enc_bits;
	std::vector<float> symbols;
	std::vector<float> sigma;
	std::vector<float> noisy_symbols;
	std::vector<float> LLRs;
	std::vector<int  > dec_bits;
};

struct utils
{
	std::unique_ptr<tools::Sigma<>>               noise;     // a sigma noise type
	std::vector<std::unique_ptr<tools::Reporter>> reporters; // list of reporters dispayed in the terminal
	std::unique_ptr<tools::Terminal_std>          terminal;  // manage the output text in the terminal
};

void init_params(params &p, cxxopts::ParseResult &opts)
{
	p.K         = opts["k"].as<int> ();
	p.N         = opts["n"].as<int> ();
	p.fe        = opts["fe"].as<int> ();
	p.seed      = opts["seed"].as<int> ();
	p.ebn0_min  = opts["ebn0min"].as<float> ();
	p.ebn0_max  = opts["ebn0max"].as<float> ();
	p.ebn0_step = opts["ebn0step"].as<float> ();
	if (opts.count("r"))
	{
		p.R = opts["r"].as<float> ();
		p.K = (int) (p.N * 1.0 * p.R);
	}
	else
	{
		p.R = (float)p.K / (float)p.N;
[... 152 lines cut ...]
		if (u.terminal->is_over()) break;
	}

	std::cout << "# End of the simulation" << std::endl;

	return 0;
}
```
Please tell me why it gives the following error, how I should fix it:
**using the AFF3CT library (v2.3.5), output:**
``` shell
terminate called after throwing an instance of 'aff3ct::tools::length_error'
  what():  In the '/home/cm/Code/my_project_with_aff3ct/lib/aff3ct/src/Module/Encoder/Encoder.hxx' file at line 145 ('encode' function): "'U_K.size()' has to be equal to 'K' * 'n_frames' ('U_K.size()' = 400, 'K' = 3600, 'n_frames' = 1)."
Backtrace:
./bin/RS_codec_test: void aff3ct::module::Encoder<int>::encode<std::allocator<int> >(std::vector<int, std::allocator<int> > const&, std::vector<int, std::allocator<int> >&, int) +0x1dc [0x56192b38e28a]
./bin/RS_codec_test: main() +0x1b1d [0x56192b37e46b]
/lib/x86_64-linux-gnu/libc.so.6: () +0x29d90 [0x7f272de29d90]
/lib/x86_64-linux-gnu/libc.so.6: __libc_start_main() +0x80 [0x7f272de29e40]
./bin/RS_codec_test: _start() +0x25 [0x56192b37a915]
Aborted (core dumped)
```
**using the AFF3CT library (v3.0.2), similar error occur in `Socket.hxx`:**
``` shell
'vector.size()' has to be equal to 'get_n_elmts()'
```

Volshebnik-Wisard, 2024-03-23:
Hello, can I see your code, otherwise my Reed-Solomon encoder (Encoder_RS) causes an unhandled Run time exception.
[image]

Volshebnik-Wisard, 2024-03-23:
> Hello, can I see your code? otherwise my Reed-Solomon encoder (Encoder_RS) causes an unhandled Run time exception. ![123](https://private-user-images.githubusercontent.com/164634663/316241819-c9247614-3b96-4a01-a43c-95e59ef84b9c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTExOTg1NjYsIm5iZiI6MTcxMTE5ODI2NiwicGF0aCI6Ii8xNjQ2MzQ2NjMvMzE2MjQxODE5LWM5MjQ3NjE0LTNiOTYtNGEwMS1hNDNjLTk1ZTU5ZWY4NGI5Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyM1QxMjUxMDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMGM3ZGIwN2Y2ZWJjOGQ5ZTY5YjJhZGQyMDllMDdjZTA3ZGY1N2VhM2QxZmRiYzNlMDFmMmIwMDBiMWQ4OTQ3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5SLjFgPWDlHainn95WIrn-nitz1FDi9GuV4LeozR83Y)

Volshebnik-Wisard, 2024-03-23:
Hello, can I see your code? Otherwise my Reed-Solomon encoder (Encoder_RS) causes an unhandled Run time exception.
[image]