aff3ct/aff3ct #189: Contribution legals: GRAND decoder

RRoHES, 2024-02-15:
Bonjour @kouchy et al.
I would like to contribute to aff3ct by providing code for Guessing Random Additive Noise Decoding (GRAND) and Ordered reliability bits guessing random additive noise decoding (ORBGRAND) implementations to integrate into aff3ct's dev branch.
The code is ready for a pull request. It works through the command line and with aff3ct's projects. Its performance was compared to the reference code available publicly.

However the reference code is subject to an "Academic and/or non-profit Internal Research End User Agreement" which I'm not sure if aff3ct is totally compatible with. EUA available here: https://github.com/kenrduffy/GRAND-MATLAB

Can you tell me if you think this would be possible to integrate this code with the proper citations/references to comply with this EUA as you probably had to do it previously?

The code is my own for my master thesis but was inspired in part from the papers and this repository.
I can take care of asking MIT Technology Licensing Office if you think it's needed.

Thanks

kouchy, 2024-02-16:
Hi @RRoHES,

First, thank you for your interest and work with AFF3CT!

I'm not a specialist to legal stuff so take what I'm saying here as a non-expert comment.

I think that the MIT license from AFF3CT is much more permissive for industrial actors compared to the license of GRAND-MATLAB. In this way they does not look very compatible. However, we are speaking about your code and not the code from this repository if I understand well. If you redeveloped everything  and even changed the language from MATLAB to C++, I think now it is your code.

In any cases, I strongly suggest you to ask the same question to the main contributors of the GRAND-MATLAB repository.

I hope it helps.

RRoHES, 2024-02-16:
Thanks for the quick reply! It's a bit more clear now.

I'll get into contact with @kenrduffy to ensure it's not against their policy and can be uploaded here with the proper references.

closed 2024-02-16 by RRoHES