aff3ct/aff3ct #190: comments from Anaconda

Phillip-M-Feldman, 2024-02-23:
I recently asked Anaconda to build this package.  They did, and provided the following comments that may be helpful:

1.	The build process uses non-pythonic processes.  It uses Doxygen to generate the list of functions to wrap in python! 
2.	The code has not been updated in 2 years.  With Python and Numpy rapidly deprecating older versions, it might not be able to built with more recent versions of python.
3.	I've updated the matplotlib dependency which was hard pinned to an old version -- it's a minor version change, so it probably won't have any breakage.

kouchy, 2024-02-24:
Hi @Phillip-M-Feldman,

If I understand well, you asked for adding `py_aff3ct` to Anaconda isn't it?
If yes, it is true that this version of the wrapper is now deprecated. To be honest we are currently working on a new version that should be released soon. We plan to distribute it over PyPi. Thus, the build process is  definitively more Pythonic.

Phillip-M-Feldman, 2024-02-24:
This is good to hear.  Thanks!