aff3ct/aff3ct #199: Codec_LDPC doesn't seem to configure correctly through factory

dwdickerhoof, 2024-11-06:
I am evaluating this library to see if it can be used to perform the 5G NR LDPC decoding.  However, I am struggling to get it configured.

I can build and run the factory example at https://github.com/aff3ct/my_project_with_aff3ct/tree/master/examples/factory.  
I then tried to modify the factory example to do LDPC instead of a repetition code:
```
struct params
{
	float ebn0_min  =  0.00f; // minimum SNR value
	float ebn0_max  = 10.01f; // maximum SNR value
	float ebn0_step =  1.00f; // SNR step
	float R;                  // code rate (R=K/N)

	std::unique_ptr<factory::Source      > source;
	std::unique_ptr<factory::Codec_LDPC  > codec;
	std::unique_ptr<factory::Modem       > modem;
	std::unique_ptr<factory::Channel     > channel;
	std::unique_ptr<factory::Monitor_BFER> monitor;
	std::unique_ptr<factory::Terminal    > terminal;
};
```
I run it with these command line parameters:
```
#!/bin/bash
bin/ldpc-5gnr \
--src-type      RAND \
--src-info-bits	2080 \
--enc-type      AZCW \
--enc-info-bits	2080 \
--enc-cw-size	4576 \
--chn-type      AWGN \
--chn-fra-size	4576 \
--mdm-type      BPSK \
--mdm-fra-size	4576 \
--dec-type      BP_HORIZONTAL_LAYERED \
--dec-h-path	/workspaces/ldpc-5g/examples/ldpc-5gnr/config/bg2-ils6-zc208.qc \
--dec-implem    NMS \
--dec-norm      0.75 \
--dec-ite       15 \
--mnt-info-bits	2080
```
Even though I have the encoder type configured for AZCW and k=2080, n=4576, the encoder seems to be going to the full width of the parity check matrix, n=10816.  Then, of course, the bind() calls fail because of size mismatches.
```
#----------------------------------------------------------
# This is a basic program using the AFF3CT library (v3.0.2)
# Feel free to improve it as you want to fit your needs.
#----------------------------------------------------------
#
# Simulation parameters: 
# * Source ----------------------------------------
#    ** Type                      = RAND
#    ** Implementation            = STD
#    ** Info. bits (K_info)       = 2080
#    ** Seed                      = 0
# * Codec -----------------------------------------
#    ** Type                      = LDPC
#    ** Info. bits (K)            = 2080
#    ** Codeword size (N_cw)      = 10816
#    ** Frame size (N)            = 10816
#    ** Code rate                 = 0.192308 (5/26)
# * Encoder ---------------------------------------
#    ** Type                      = AZCW
#    ** Info. bits (K)            = 2080
#    ** Codeword size (N)         = 10816
#    ** Code rate (R)             = 0.192308
#    ** Systematic                = yes
# * Decoder ---------------------------------------
#    ** Type (D)                  = BP_HORIZONTAL_LAYERED
#    ** Implementation            = NMS
#    ** Info. bits (K)            = 2080
#    ** Codeword size (N)         = 10816
#    ** Code rate (R)             = 0.192308
#    ** Systematic                = yes
#    ** Seed                      = 0
#    ** H matrix path             = /workspaces/ldpc-5g/examples/ldpc-5gnr/config/bg2-ils6-zc208.qc
#    ** H matrix reordering       = NONE
#    ** Num. of iterations (i)    = 15
#    ** Normalize factor          = 0.750000
#    ** Stop criterion (syndrome) = on
#    ** Stop criterion depth      = 1
# * Modem -----------------------------------------
#    ** Type                      = BPSK
#    ** Implementation            = STD
#    ** Frame size (N)            = 4576
#    ** Bits per symbol           = 1
#    ** Sigma square              = on
#    ** Channel type              = AWGN
# * Channel ---------------------------------------
#    ** Type                      = AWGN
#    ** Implementation            = STD
#    ** Frame size (N)            = 4576
#    ** Seed                      = 0
#    ** Complex                   = off
#    ** Add users                 = off
# * Monitor ---------------------------------------
#    ** Frame error count (e)     = 100
#    ** Size (K)                  = 2080
# * Terminal --------------------------------------
#    ** Enabled                   = yes
#    ** Frequency (ms)            = 500
#
# ---------------------||------------------------------------------------------||---------------------
#  Signal Noise Ratio  ||   Bit Error Rate (BER) and Frame Error Rate (FER)    ||  Global throughput  
#         (SNR)        ||                                                      ||  and elapsed time   
# ---------------------||------------------------------------------------------||---------------------
# ----------|----------||----------|----------|----------|----------|----------||----------|----------
#     Es/N0 |    Eb/N0 ||      FRA |       BE |       FE |      BER |      FER ||  SIM_THR |    ET/RT 
#      (dB) |     (dB) ||          |          |          |          |          ||   (Mb/s) | (hhmmss) 
# ----------|----------||----------|----------|----------|----------|----------||----------|----------
terminate called after throwing an instance of 'aff3ct::tools::runtime_error'
  what():  In the '/workspaces/ldpc-5g/lib/aff3ct/include/Module/Socket.hxx' file at line 174 ('bind' function): "'s_out.databytes' has to be equal to 'databytes' ('s_out.databytes' = 43264, 's_out.name' = X_N, 's_out.task.name' = encode, 'databytes' = 18304, 'name' = X_N1, 'task.name' = modulate)."
Backtrace:
bin/ldpc-5gnr: aff3ct::module::Socket::bind(aff3ct::module::Socket&, int) +0x6b2 [0x5596666a8c94]
bin/ldpc-5gnr: main() +0x402 [0x5596666a56fd]
/lib/x86_64-linux-gnu/libc.so.6: __libc_start_main() +0xf3 [0x7febb3263083]
bin/ldpc-5gnr: _start() +0x2e [0x5596666a520e]
./run5g: line 17: 48923 Aborted                 (core dumped) bin/ldpc-5gnr --src-type RAND --src-info-bits 2080 --enc-type AZCW --enc-info-bits 2080 --enc-cw-size 4576 --chn-type AWGN --chn-fra-size 4576 --mdm-type BPSK --mdm-fra-size 4576 --dec-type BP_HORIZONTAL_LAYERED --dec-h-path /workspaces/ldpc-5g/examples/ldpc-5gnr/config/bg2-ils6-zc208.qc --dec-implem NMS --dec-norm 0.75 --dec-ite 15 --mnt-info-bits 2080
```
Am I doing something wrong here, or is this a bug?

kouchy, 2024-11-06:
Hi @dwdickerhoof,
Could you share the `bg2-ils6-zc208.qc` file with us please?

dwdickerhoof, 2024-11-06:
> Caution: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Thanks for your quick response!  The parity check matrix definition is attached.

Best regards,

Dale

dwdickerhoof, 2024-11-06:
[bg2-ils6-zc208.qc.txt](https://github.com/user-attachments/files/17651068/bg2-ils6-zc208.qc.txt)

Renamed to end in .txt in order to submit.

dwdickerhoof, 2024-11-06:
If I edit the .qc file to shorten H to 12 rows and 22 columns, and also change to  --src-type AZCW, then it appears to work.
```
#----------------------------------------------------------
#
# Simulation parameters: 
# * Source ----------------------------------------
#    ** Type                      = AZCW
#    ** Implementation            = STD
#    ** Info. bits (K_info)       = 2080
# * Codec -----------------------------------------
#    ** Type                      = LDPC
#    ** Info. bits (K)            = 2080
#    ** Codeword size (N_cw)      = 4576
#    ** Frame size (N)            = 4576
#    ** Code rate                 = 0.454545 (5/11)
# * Encoder ---------------------------------------
#    ** Type                      = AZCW
#    ** Info. bits (K)            = 2080
#    ** Codeword size (N)         = 4576
#    ** Code rate (R)             = 0.454545
#    ** Systematic                = yes
# * Decoder ---------------------------------------
#    ** Type (D)                  = BP_HORIZONTAL_LAYERED
#    ** Implementation            = NMS
#    ** Info. bits (K)            = 2080
#    ** Codeword size (N)         = 4576
#    ** Code rate (R)             = 0.454545
#    ** Systematic                = yes
#    ** Seed                      = 0
#    ** H matrix path             = /workspaces/ldpc-5g/examples/ldpc-5gnr/config/bg2-ils6-zc208-short.qc
#    ** H matrix reordering       = NONE
#    ** Num. of iterations (i)    = 15
#    ** Normalize factor          = 0.750000
#    ** Stop criterion (syndrome) = on
#    ** Stop criterion depth      = 1
# * Modem -----------------------------------------
#    ** Type                      = BPSK
#    ** Implementation            = STD
#    ** Frame size (N)            = 4576
#    ** Bits per symbol           = 1
#    ** Sigma square              = on
#    ** Channel type              = AWGN
# * Channel ---------------------------------------
#    ** Type                      = AWGN
#    ** Implementation            = STD
#    ** Frame size (N)            = 4576
#    ** Seed                      = 0
#    ** Complex                   = off
#    ** Add users                 = off
# * Monitor ---------------------------------------
#    ** Frame error count (e)     = 100
#    ** Size (K)                  = 2080
# * Terminal --------------------------------------
#    ** Enabled                   = yes
#    ** Frequency (ms)            = 500
#
# ---------------------||------------------------------------------------------||---------------------
#  Signal Noise Ratio  ||   Bit Error Rate (BER) and Frame Error Rate (FER)    ||  Global throughput  
#         (SNR)        ||                                                      ||  and elapsed time   
# ---------------------||------------------------------------------------------||---------------------
# ----------|----------||----------|----------|----------|----------|----------||----------|----------
#     Es/N0 |    Eb/N0 ||      FRA |       BE |       FE |      BER |      FER ||  SIM_THR |    ET/RT 
#      (dB) |     (dB) ||          |          |          |          |          ||   (Mb/s) | (hhmmss) 
# ----------|----------||----------|----------|----------|----------|----------||----------|----------
      -3.42 |     0.00 ||      100 |    29545 |      100 | 1.42e-01 | 1.00e+00 ||    0.118 | 00h00'01   
      -3.17 |     0.25 ||      100 |    25203 |      100 | 1.21e-01 | 1.00e+00 ||    0.119 | 00h00'01   
      -2.92 |     0.50 ||      100 |    19262 |      100 | 9.26e-02 | 1.00e+00 ||    0.120 | 00h00'01   
      -2.67 |     0.75 ||      111 |    11073 |      100 | 4.80e-02 | 9.01e-01 ||    0.121 | 00h00'01   
      -2.42 |     1.00 ||      232 |     5031 |      100 | 1.04e-02 | 4.31e-01 ||    0.136 | 00h00'03   
      -2.17 |     1.25 ||     1284 |     2800 |      100 | 1.05e-03 | 7.79e-02 ||    0.165 | 00h00'16   
      -1.92 |     1.50 ||    25149 |     1628 |      100 | 3.11e-05 | 3.98e-03 ||    0.210 | 00h04'08   
      -1.67 |     1.75 ||    51068 |       85 |        9 | 8.00e-07 | 1.76e-04 ||    0.257 | 00h06'52  x

```