aff3ct/aff3ct #203: What is needed to use --dec-simd options?

enyg, 2025-04-17:
I have tried using the `--dec-simd INTER` and `--dec-simd INTRA` options, and I get errors like the following:

```
(EE) In the '<path>/aff3ct/src/Factory/Module/Decoder/LDPC/Decoder_LDPC.cpp' file at line 379 ('build_siso' function): "Cannot allocate the object."
(EE) Backtrace:
(EE) <path>/aff3ct/build_amd/bin/aff3ct-3.0.2: aff3ct::module::Decoder_SISO<short, short>* aff3ct::factory::Decoder_LDPC::build_siso<short, short>(aff3ct::tools::Sparse_matrix const&, std::vector<unsigned int, std::allocator<unsigned int> > const&, aff3ct::module::Encoder<short>*) const +0x59f [0x55bacbea328f]
(EE) <path>/aff3ct/build_amd/bin/aff3ct-3.0.2: aff3ct::module::Decoder_SIHO<short, short>* aff3ct::factory::Decoder_LDPC::build<short, short>(aff3ct::tools::Sparse_matrix const&, std::vector<unsigned int, std::allocator<unsigned int> > const&, aff3ct::module::Encoder<short>*) const +0x199 [0x55bacbea42c9]
(EE) <path>/aff3ct/build_amd/bin/aff3ct-3.0.2: aff3ct::tools::Codec_LDPC<short, short>::Codec_LDPC(aff3ct::factory::Encoder_LDPC const&, aff3ct::factory::Decoder_LDPC const&, aff3ct::factory::Puncturer_LDPC*) +0xc37 [0x55baccddbe67]
(EE) <path>/aff3ct/build_amd/bin/aff3ct-3.0.2: aff3ct::tools::Codec_LDPC<short, short>* aff3ct::factory::Codec_LDPC::build<short, short>(aff3ct::module::CRC<short> const*) const +0x8e [0x55bacc588a4e]
(EE) <path>/aff3ct/build_amd/bin/aff3ct-3.0.2: aff3ct::tools::Codec_SIHO<short, short>* aff3ct::factory::Codec_SIHO::build<short, short>(aff3ct::module::CRC<short> const*) const +0x105 [0x55bacc57c625]
(EE) <path>/aff3ct/build_amd/bin/aff3ct-3.0.2: aff3ct::simulation::Simulation_BFER_std<short, float, short>::build_codec(aff3ct::module::CRC<short> const*) +0x77 [0x55baccccae87]
(EE) <path>/aff3ct/build_amd/bin/aff3ct-3.0.2: aff3ct::simulation::Simulation_BFER_std<short, float, short>::create_modules() +0xa3 [0x55baccccb383]
(EE) <path>/aff3ct/build_amd/bin/aff3ct-3.0.2: aff3ct::simulation::Simulation_BFER<short, float>::launch() +0x38 [0x55bacccbc308]
(EE) <path>/aff3ct/build_amd/bin/aff3ct-3.0.2: aff3ct::launcher::Launcher::launch() +0x156 [0x55bacc688bc6]
(EE) <path>/aff3ct/build_amd/bin/aff3ct-3.0.2: main() +0xc8 [0x55bacbe266d8]
(EE) /lib64/libc.so.6: __libc_start_main() +0xf5 [0x7fbe8fb58555]
(EE) <path>/aff3ct/build_amd/bin/aff3ct-3.0.2: () +0x5ec9f7 [0x55bacbe269f7]
```

This was on an executable I built from source (tag 3.0.2 from github) following the compilation instructions. I have also had similar errors with the prebuilt windows executables. What do I need to do to get the SIMD options to work?

Thanks

bonben, 2025-04-18:
Hello Erich.

SIMD is not available for every LDPC decoder implem.
Compatibilities are listed [there](https://aff3ct.readthedocs.io/en/develop/user/simulation/parameters/codec/ldpc/decoder.html#tab-ldpc-dec-implem).

Regards,
Mathieu.

enyg, 2025-04-21:
I am using decoder implementations that are compatible with SIMD according to the table you linked to.

examples of some configurations I have tried:
`--dec-type BP_HORIZONTAL_LAYERED --dec-implem SPA --sim-prec 32`
`--dec-type BP_HORIZONTAL_LAYERED --dec-implem AMS --dec-implem MINS --sim-prec 16 --qunt-bits 7 --qnt-dec 2`
`--dec-type BP_HORIZONTAL_LAYERED --dec-implem AMS --dec-implem MINS --sim-prec 32`
`--dec-type BP_HORIZONTAL_LAYERED --dec-implem AMS --dec-implem MINL --sim-prec 32`
`--dec-type BP_VERTICAL_LAYERED --dec-implem LSPA --sim-prec 16 --qunt-bits 7 --qnt-dec 2`

I am using `--enc-type AZCW` also. All of the above were tried with and without puncturing, and with `--dec-simd INTER` and `INTRA`. The result is always the same ("Cannot allocate the object" as shown in the original message). So I don't think the issue is that I'm using an algorithm that isn't SIMD compatible, unless the documentation is inaccurate. Any other ideas?

enyg, 2025-04-21:
I had not compiled with `-faligned-new` yet. I just tried it, and INTER is working for me now. Thanks!

I guess I missed that detail in the documentation, even though it is there. It could probably be stated more clearly that the decoder implementations with ** need the `-faligned-new` option specifically for SIMD. The way it is written now, it makes it seem like those decoder implementations need that option in general. (I may have ignored it because I thought the option was enabled by default, since I could use those decoders in some cases.)

Perhaps even more helpful would be to include on the [Compilation](https://aff3ct.readthedocs.io/en/latest/user/installation/compilation/compilation.html) page a list of the compiler flags and what AFF3CT features they are required for.